salamando15/TileBot #21: Cesar - Review APDS 9960 Part and related engineering material/principles

Zoom9000, 2023-02-19:
Preliminary:
I2C protocal software and hardware connectivity
Connect to Pi and ensure compatibility and usability
Check for ambient light, proximity, red, blue, green, and gesture readings

ffund, 2023-03-01:
Since the code in 62202eb does not seem to get correct readings, we will try using the [Adafruit library](https://github.com/adafruit/Adafruit_CircuitPython_TCS34725).

Step 1:

```
sudo pip3 install adafruit-circuitpython-tcs34725
```

Step 2:

get report and run the example

```
git clone https://github.com/adafruit/Adafruit_CircuitPython_TCS34725
cd Adafruit_CircuitPython_TCS34725/examples
python3 tcs34725_simpletest.py
```

ffund, 2023-03-01:
We tested with a few different colors and can definitely see the R/G/B value increase according to expectation.

Turquoise:
[image]

[image]

Red: 
[image]
[image]

Blue:
[image]
[image]

Purple:
[image]
[image]

ffund, 2023-03-01:
Note: this was without interrupt pin.

Zoom9000, 2023-05-15:
Red: (59,5,5) or (79,8,8)

white:  (14,18,14)

Blue: (16,16,16)  (9,16,25)

closed 2023-05-16 by salamando15