salamando15/TileBot #29: Scematic for wiring RFID reader

CarlosLopezGit, 2023-03-01:
- [ ] create schematic for wiring RFID reader to the pi

- Make sure there are no conflicts with existing schematics we made
- If a conflict arises contact the team member who's part conflicts and resolve conflict if possible, make note of this conflict in the comment for this issue

CarlosLopezGit, 2023-03-08:
[image]

CarlosLopezGit, 2023-03-28:
change i2c bus to pins 2 and 3

Zoom9000, 2023-04-28:
Tried to connect using I2C but no helpful resources were found online or practical implementations of I2C using the RFID 
Spoke to TA about implementing it in I2C but advised to use SPI 

[image]

closed 2023-05-16 by salamando15